halajun/VDO_SLAM #1: error: ‘SOLVEPNP_AP3P’ is not a member of ‘cv’

haleqiu, 2020-07-02:
Hi Jun,
I am following your instruction in `build.sh`. I met a opencv problem as followed:
```
/home/yuheng/project/VDO_SLAM/src/Tracking.cc:1752:70: error: ‘SOLVEPNP_AP3P’ is not a member of ‘cv’
                iter_num, reprojectionError, confidence, inliers, cv::SOLVEPNP_AP3P); // AP3P EPNP P3P ITERATIVE DLS
                                                                      ^~~~~~~~~~~~~
/home/yuheng/project/VDO_SLAM/src/Tracking.cc:1752:70: note: suggested alternative: ‘SOLVEPNP_P3P’
                iter_num, reprojectionError, confidence, inliers, cv::SOLVEPNP_AP3P); // AP3P EPNP P3P ITERATIVE DLS
                                                                      ^~~~~~~~~~~~~
                                                                      SOLVEPNP_P3P
```
I am using Opencv 3.2. Is there any suggestions for this issue.

halajun, 2020-07-02:
Hi, @haleqiu.  I tested with opencv 3.4.  I guess the 3.2 version does not contain the AP3P method. You can use the standard P3P as suggested by your compiler. It should produce similar results, and the difference might be just efficiency. Hope this helps.

MinaHenein, 2020-07-21:
@haleqiu Does that solve your issue? If so, I will close this issue.

closed 2020-08-04 by MinaHenein