halajun/VDO_SLAM #8: What's the difference between "mvStatKeysTmp" and "mvStatKeys"?

ShaotengWu, 2020-12-03:
Hi halajun! Thanks for your great opening source work. I am interested in your coding details. I am currently stuck by the meaning of some variables as the title. Sorry for interrupting and would you give me a hand?

halajun, 2020-12-04:
Hi, @ShaotengWu. Thanks for the interest in VDO-SLAM. The suffix term "Tmp" is I use in the code refers to temporary variable. Specifically, for each new coming frame, the variable "mvStatKeysTmp" is the detected or sampled keypoints in the new frame. But for current frame, in fact, we use the correspondences from last frame as the keypoints in current frame, i.e., mvStatKeys (see line 251 in Tracking.cc) to keep tracking of the same points continuously over frames. Please be aware that the optical flow in current frame we used, represent the correspondences between current frame and the next frame, not between current frame and last frame. Hope this helps.

ShaotengWu, 2020-12-05:
Thank you a lot!

closed 2020-12-05 by ShaotengWu