halajun/VDO_SLAM #11: How to solve this problem，thank you

xiaohang212, 2021-03-02:
[100%] Linking CXX executable ../example/vdo_slam
CMakeFiles/vdo_slam.dir/example/vdo_slam.cc.o: In function `main':
vdo_slam.cc:(.text.startup+0xc08): undefined reference to `cv::optflow::readOpticalFlow(cv::String const&)'
collect2: error: ld returned 1 exit status
CMakeFiles/vdo_slam.dir/build.make:115: recipe for target '../example/vdo_slam' failed
make[2]: *** [../example/vdo_slam] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/vdo_slam.dir/all' failed
make[1]: *** [CMakeFiles/vdo_slam.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

halajun, 2021-03-02:
Hi, @xiaohang212. Looks like it has something to do with the function of "cv::optflow::readOpticalFlow()"; Please check your opencv version if it support this function, shouldn't be a difficult issue to fix. Hope this helps.

MinaHenein, 2021-03-09:
Hi @xiaohang212,
Please let us know if you solved this issue.

xiaohang212, 2021-03-13:
Problem solved, thank you～

closed 2021-03-16 by halajun

Duke1325, 2021-09-08:
@xiaohang212 
I met the same issue, could you share the way to solve it? Thanks!

colin-de, 2022-08-29:
@xiaohang212 
hello xiaohang, could u please tell me how did u solve this issue?
I still have the problem as you before.
Thanks! :)

Duke1325, 2022-08-30:
您的邮件我已经收到，谢谢！

Duke1325, 2022-10-11:
您的邮件我已经收到，谢谢！