halajun/VDO_SLAM #14: Issue in PlainObjectBase.h When Running Demos

jingGM, 2021-03-23:
The compliation is fine. But when I run the command to run the demo datasets, there is an issue: 

vdo_slam: /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:285: void Eigen::PlainObjectBase<Derived>::resize(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<double, 3, 3>; Eigen::Index = long int]: Assertion `(!(RowsAtCompileTime!=Dynamic) || (rows==RowsAtCompileTime)) && (!(ColsAtCompileTime!=Dynamic) || (cols==ColsAtCompileTime)) && (!(RowsAtCompileTime==Dynamic && MaxRowsAtCompileTime!=Dynamic) || (rows<=MaxRowsAtCompileTime)) && (!(ColsAtCompileTime==Dynamic && MaxColsAtCompileTime!=Dynamic) || (cols<=MaxColsAtCompileTime)) && rows>=0 && cols>=0 && "Invalid sizes when resizing a matrix or array."' failed.
Aborted (core dumped)

halajun, 2021-03-31:
Hi, @jingGM . Did you change any matrix in the code? I checked this online, it is all related to false matrix size. If not, I think maybe this issue is due to the Eigen version, probably in newer version, the function that causes the issue has changed. Hope this helps.

closed 2021-04-17 by halajun

xuehu2, 2023-10-24:
> Hi, @jingGM . Did you change any matrix in the code? I checked this online, it is all related to false matrix size. If not, I think maybe this issue is due to the Eigen version, probably in newer version, the function that causes the issue has changed. Hope this helps.

Hi, @halajun . I meet this issue. but there is a thing that I run the program in terminal, it works fine, no issue pop up. But when I try to run the program in Clion IDE, this issue pop up. Now I can not point the reason of this issue.

If anybody meet this in an IDE, try Terminal. But it make you did not Debug the code...... :(

xuehu2, 2023-10-24:
https://github.com/halajun/VDO_SLAM/issues/18#issuecomment-864636236
Hi @halajun . I found a issue that u quote cannot run in debug mode. I tried it, it do not run successfully in debug mode. So do u have any method to solve this issue in debug mode?