halajun/VDO_SLAM #15: How to get the Ground Truth Camera Pose , Object Pose and the times.txt

Liuxuweilxw, 2021-03-27:
I may can get the bounding box (B1-B4) from Mask-RCNN  ,but I don't know how to get others data. And there is no times.txt  in KITTI tracking dataset.  Looking forward to your reply. Thanks.

halajun, 2021-03-31:
Hi, @Liuxuweilxw. The ground truth camera pose is in the "data_tracking_oxts" folder if you have downloaded it from the KITTI Tracking Dataset. the files inside saves the GPS/IMU data, where you can extract the ground truth camera pose. For the ground truth object pose, it is in the "data_tracking_label_2" if you have downloaded. Please refer to the readme file in this dataset for more details. Finally for the "times.txt", it is just used for organizing the image frames, you can generate it yourself with any frame rate (e.g., 10hz), it does not affect anything in the code. Hope this helps.

Liuxuweilxw, 2021-04-01:
Thanks for your help !

closed 2021-04-08 by halajun