halajun/VDO_SLAM #16: how to generate the depth map?

shqfeng, 2021-03-28:
I try to use SGBM to calculate the depth, but it dose not work! Can you share the project of depth map generation?

Liuxuweilxw, 2021-03-29:
Try monodepth2 https://github.com/nianticlabs/monodepth2

Liuxuweilxw, 2021-03-29:
It's for monocular

shqfeng, 2021-03-29:
thanks for your reply！ how to generate the "depth" folder of KITTI stereo dataset, I use the opencv SGBM, hower, the results is bad, can you share this work? thanks very much!

Liuxuweilxw, 2021-03-29:
Sorry, I make wrong. I used monodepth to generate depth for single image, not for stereo

halajun, 2021-03-31:
Hi, @shqfeng. the method SGBM generates disparity map, not depth map. You need to compute the depth value from disparity by [depth = baseline*focal_lenth/(disparity/depth_factor)]. For your reference, I use the [method](https://ttic.uchicago.edu/~dmcallester/SPS/index.html) of the paper "Efficient Joint Segmentation, Occlusion Labeling, Stereo and Flow Estimation" to calculate the disparity map. Hope this helps.

closed 2021-04-08 by halajun