halajun/VDO_SLAM #18: 如何关联当前帧MaskRCNN提取的对象与上一帧的同一对象？

huxiao987, 2021-05-17:
张博士你好，我在跑完您的VDO-SLAM的代码后，有个疑问，您是如何关联每一帧MaskRCNN提起出来的对象的？比如说如何把当前帧提取出的汽车关联到上一帧同一辆汽车，在代码中哪里有提现么？谢谢！

MinaHenein, 2021-05-17:
Hi @huxiao987,
Thanks a lot for your question. 
To answer you question, we use optical flow for the tracking of different objects across frames. 
This is explained in more details in the "Tracking" section in our paper
https://arxiv.org/pdf/2005.11052.pdf

huxiao987, 2021-05-17:
thanks for your answer，when I run VDO SLAM by debug ,program will occurs this error  

vdo_slam: /usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:285：void Eigen::PlainObjectBase<Derived>::resize(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<double, 3, 3>; Eigen::Index = long int]: 假设 ‘(!(RowsAtCompileTime!=Dynamic) || (rows==RowsAtCompileTime)) && (!(ColsAtCompileTime!=Dynamic) || (cols==ColsAtCompileTime)) && (!(RowsAtCompileTime==Dynamic && MaxRowsAtCompileTime!=Dynamic) || (rows<=MaxRowsAtCompileTime)) && (!(ColsAtCompileTime==Dynamic && MaxColsAtCompileTime!=Dynamic) || (cols<=MaxColsAtCompileTime)) && rows>=0 && cols>=0 && "Invalid sizes when resizing a matrix or array."’ 失败。

when I run by Release ,progrem can run.Thank you.

halajun, 2021-06-21:
Hi @huxiao987 , sorry for the late reply. Have you solved the problem? looks like it has something to do with the Eigen library being not recognized in the debug mode?

closed 2021-12-17 by MinaHenein