halajun/VDO_SLAM #19: Semantic Segmentation of the boxes from the Oxford Multimotion Dataset

JavierTG97, 2021-05-20:
Amazing work :).
I just have a question regarding the semantic segmentation of the boxes from the Oxford Multimotion Dataset. How is it done?
I am confused because COCO dataset doesn't have a labels like "cubes" or "boxes".
Thanks in advance!

halajun, 2021-06-21:
Hi @JavierTG97. Thanks for your interest, and apologize for the late reply. You are pointing out a very good issue here, where we would correct in later version. For the segmentation of Oxford dataset, we don't use learning method to infer the mask. Instead, we use Otsu's method, together with color information and multi-label processing to segment the boxes, which works very well for the simple setup of this dataset (color boxes that are highly distinguishable from the background).

JavierTG97, 2021-06-21:
@halajun thanks a lot! :) That solves my question.

closed 2021-06-21 by JavierTG97