halajun/VDO_SLAM #20: What is your input image to generate the .flo file with PWC-NET?

real-chenzy, 2021-05-29:
I use pytorch-pwc to generate the .flo file but it seems to be no data within the object,
Firstly, the input to pwc I used is the original image, 
[image]
after I generate the .flo file, I test the vdo_slam, but it seems no data in flow
[image]
[image]
then I convert it to .png file, but it seems to be no flow in the object:
[image]

real-chenzy, 2021-05-29:
After I comment these lines, it only work on first 2 frames:
[image]
[image]

and it can't detect the moving car in the image

halajun, 2021-06-20:
Hi, @real-chenzy. Sorry for the late reply. I think the mask of object has been filtered in the tracking process, due to the object being too small in the scene. You can change the pre-set variable of "ThDepthOBJ" in the yaml file, or modify the thresholds in the function of Tracking::DynObjTracking(). Hope this may help.