halajun/VDO_SLAM #22: run the example in Figure 9 of your paper

hanpengpeng123, 2021-06-25:
Thank you for the open source code.
 I want to run the example in Figure 9 of your paper. Can you give me the relevant files? My email is 1421384659@qq.com, thank you very much

halajun, 2021-07-08:
Hi, @hanpengpeng123. Thanks for your interest. The complete KITTI data files are around 20GB in size (mainly the optic-flow and semantic mask takes up a lot of storage), which is difficult for me to share with you online. May I ask what issues you have encountered when trying to generate the data yourself? I am happy to help you with that.

hanpengpeng123, 2021-07-08:
> 嗨，@hanpengpeng123。谢谢你的关注。完整的 KITTI 数据文件大小在 20GB 左右（主要是光流和语义掩码占用了大量存储空间），我很难在线与大家分享。请问您在尝试自己生成数据时遇到了哪些问题？我很乐意为您提供帮助。
Can files other than optic-flow and semantic mask be shared?

andersonhusky, 2022-07-03:
> Hi, @hanpengpeng123. Thanks for your interest. The complete KITTI data files are around 20GB in size (mainly the optic-flow and semantic mask takes up a lot of storage), which is difficult for me to [...]

i really wonder how did you get the depth information, it's a important part of the input data but i just don't have a proper solution, thx!