halajun/VDO_SLAM #23: quertion

9528-ren, 2021-07-05:
Hi！@halajun 
How to export TXT document from data set trained by mask R-CNN and Pw net
looking forward your reply

hanpengpeng123, 2021-07-06:
> 你好！@halajun
> 如何从mask R-CNN和Pw net训练的数据集中导出TXT文档
> 期待你的回复

Brother, if you know, can you share it with me？I want to run the complete kitti data set, thank you very much

halajun, 2021-07-08:
Hi @9528-ren. 

(1) For the txt document from Mask R-CNN: as the default output from Mask R-CNN is a binary mask (current image size) for each detected object, so I assign each detected object with new label, i.e., 1,2,...,n for all objects detected in current image and 0 for background, and save it as a txt file of image size. 

(2) the output of PWC-NET is not txt file, but .flo file, which is standard optical flow file format, and can be parsed directly using opencv.

Hope this helps.

closed 2021-08-11 by MinaHenein