halajun/VDO_SLAM #24: Why are the number of .flo files and .png files the same?

Summer2474, 2021-07-28:
Sorry, my question may be a little naive, but I really don't understand it. Optical flow is the result of two consecutive frames, so the number of .flo should be 1 less than the number of .png. Why are the number of .flo and .png files the same in the kitti-demo provided? If I understand correctly, does 000000.flo correspond to 000000.png-000001.png? So which two pictures correspond to the last 000055.flo?

MinaHenein, 2021-08-11:
Not a naive question, it makes perfect sense. Even though the number of .flo files and .png files are the same, I'm pretty sure the implementation uses 1 less .flo file. @halajun, can you please confirm?

halajun, 2022-01-08:
Hi @Summer2474. You are right about it. I shouldn't add the 000055.flo into the demo folder, as it corresponds to the flow between image 000055.png and 0000056.png (which is not included in the demo folder). Thanks for pointing this out.

JonnyG123789, 2022-12-10:
Thanks for your sense question!