halajun/VDO_SLAM #27: make g2o several error

hongu0603, 2021-09-06:
when i make g2o , i get several error about it 

`  from /home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/edge_se3_prior.cpp:27:
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:38:3: error: expected unqualified-id before ‘using’
   using VectorN = Eigen::Matrix<number_t, N, 1, Eigen::ColMajor>;
   ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:39:9: error: expected nested-name-specifier before ‘Vector2’
   using Vector2 = VectorN<2>;
         ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:40:9: error: expected nested-name-specifier before ‘Vector3’
   using Vector3 = VectorN<3>;
         ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:41:9: error: expected nested-name-specifier before ‘Vector4’
   using Vector4 = VectorN<4>;
         ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:42:9: error: expected nested-name-specifier before ‘Vector6’
   using Vector6 = VectorN<6>;
         ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:43:9: error: expected nested-name-specifier before ‘Vector7’
   using Vector7 = VectorN<7>;
         ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:44:9: error: expected nested-name-specifier before ‘VectorX’
   using VectorX = VectorN<Eigen::Dynamic>;
         ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:47:3: error: expected unqualified-id before ‘using’
   using MatrixN = Eigen::Matrix<number_t, N, N, Eigen::ColMajor>;
   ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:48:9: error: expected nested-name-specifier before ‘Matrix2’
   using Matrix2 = MatrixN<2>;
         ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:49:9: error: expected nested-name-specifier before ‘Matrix3’
   using Matrix3 = MatrixN<3>;
         ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:50:9: error: expected nested-name-specifier before ‘Matrix4’
   using Matrix4 = MatrixN<4>;
         ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:51:9: error: expected nested-name-specifier before ‘MatrixX’
   using MatrixX = MatrixN<Eigen::Dynamic>;
         ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:53:28: error: ‘number_t’ was not declared in this scope
   typedef Eigen::Transform<number_t,2,Eigen::Isometry,Eigen::ColMajor>            Isometry2;
                            ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:53:70: error: template argument 1 is invalid
   typedef Eigen::Transform<number_t,2,Eigen::Isometry,Eigen::ColMajor>            Isometry2;
                                                                      ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:54:28: error: ‘number_t’ was not declared in this scope
   typedef Eigen::Transform<number_t,3,Eigen::Isometry,Eigen::ColMajor>            Isometry3;
                            ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:54:70: error: template argument 1 is invalid
   typedef Eigen::Transform<number_t,3,Eigen::Isometry,Eigen::ColMajor>            Isometry3;
                                                                      ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:56:28: error: ‘number_t’ was not declared in this scope
   typedef Eigen::Transform<number_t,2,Eigen::Affine,Eigen::ColMajor>              Affine2;
                            ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:56:68: error: template argument 1 is invalid
   typedef Eigen::Transform<number_t,2,Eigen::Affine,Eigen::ColMajor>              Affine2;
                                                                    ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:57:28: error: ‘number_t’ was not declared in this scope
   typedef Eigen::Transform<number_t,3,Eigen::Affine,Eigen::ColMajor>              Affine3;
                            ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:57:68: error: template argument 1 is invalid
   typedef Eigen::Transform<number_t,3,Eigen::Affine,Eigen::ColMajor>              Affine3;
                                                                    ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:59:29: error: ‘number_t’ was not declared in this scope
   typedef Eigen::Rotation2D<number_t>                                             Rotation2D;
                             ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:59:37: error: template argument 1 is invalid
   typedef Eigen::Rotation2D<number_t>                                             Rotation2D;
                                     ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/eigen_types_new.h:61:29: error: ‘number_t’ was not declared in this scope
[... 185 lines cut ...]
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/vertex_se3.h: In member function ‘virtual bool g2o::VertexSE3::getMinimalEstimateData(int*) const’:
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/vertex_se3.h:89:20: error: ‘Vector6’ was not declared in this scope
         Eigen::Map<Vector6> v(est);
                    ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/vertex_se3.h:89:27: error: template argument 1 is invalid
         Eigen::Map<Vector6> v(est);
                           ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/vertex_se3.h:89:34: error: invalid conversion from ‘int*’ to ‘int’ [-fpermissive]
         Eigen::Map<Vector6> v(est);
                                  ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/vertex_se3.h:90:13: error: ‘toVectorMQT’ is not a member of ‘g2o::internal’
         v = internal::toVectorMQT(_estimate);
             ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/vertex_se3.h: In member function ‘virtual void g2o::VertexSE3::oplusImpl(const int*)’:
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/vertex_se3.h:107:26: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
         Eigen::Map<const Vector6> v(update);
                          ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/vertex_se3.h:107:33: error: template argument 1 is invalid
         Eigen::Map<const Vector6> v(update);
                                 ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/vertex_se3.h:107:43: error: invalid conversion from ‘const int*’ to ‘int’ [-fpermissive]
         Eigen::Map<const Vector6> v(update);
                                           ^
/home/henry/henry_ws/src/VDO_SLAM/dependencies/g2o/g2o/types/vertex_se3.h:112:66: error: request for member ‘matrix’ in ‘((g2o::VertexSE3*)this)->g2o::VertexSE3::<anonymous>.g2o::BaseVertex<6, int>::_estimate’, which is of non-class type ‘g2o::BaseVertex<6, int>::EstimateType {aka int}’
           internal::approximateNearestOrthogonalMatrix(_estimate.matrix().topLeftCorner<3,3>());`

halajun, 2022-01-08:
Hi @hongu0603, sorry for the long late reply. Seems it is related to Eigen library. Please make sure you have installed the right version (>=3.1.0). Anyway, hope you have solved it earlier. Cheers.

PengSO, 2022-05-07:
add  this code in CMakeLists of g2o:
`SET(CMAKE_CXX_FLAGS "-std=c++11")  # Add c++11 functionality`
then run code
`cmake ..`