halajun/VDO_SLAM #33: Output of segmentation mask from Mask-RCNN

Rushm1an, 2021-11-23:
Hi,

Thanks for this wonderful repository of VDO-SLAM. I am trying to use Mask-RCNN directly into this system, instead of using the semantic inputs. The mask matrix that I am getting as output gives the same coefficient value for both the cars in the KITTI dataset. I even used the "kitti_mask_sem2gt.cpp" file's code to make the Mask-RCNN's output mask matrix similar to the Ground Truth object pose. 

The issue is that the two objects (cars) have the same shade of color as shown in the figure below. I want a similar output as yours. How do I solve this?

[image]

halajun, 2022-01-08:
Hi @Rushmian. According to your description, your direct output from the Mask R-CNN gives the same label to both cars, which means it only tells semantic label (i.e., they are car), while does not distinguish the instance (i.e., let's say car 1 and car 2). I guess that's why they are shown as the same color. Note that, the input into VDO-SLAM is semantic INSTANCE segmentation label. Hope this helps. Cheers.

PengSO, 2022-08-03:
@Rushmian Hi！I got the same problem，would you mind telling me have you solved your problem？

RGthesis, 2022-09-13:
Hi everyone ! I am a begginer in this project and I would love to understand and be able to work on VDO-SLAM. But I still don't understand how to get the semantic .txt file as an output of Mask-RCNN. Maybe I don't understand a part of the process ... It would be very helpful if you could help me understand this... Thank you !