halajun/VDO_SLAM #34: How to get pose_ gt. txt data

pengzhongcui, 2021-12-14:
Kitti's tracking data set only has the pose of the object, not the pose of the camera. How do you get the real pose of the camera? I want to try other data, not just those given in "kitti_demo"

halajun, 2022-01-08:
Hi @pengzhongcui. The pose of camera is from KITTI raw data, of the file/folder name "oxts", which use the gps as ground truth.

SweetZed, 2023-07-05:
> Owner
if:
A = Tr_velo_to_cam*Tr_imu_to_velo;
B = Tr_cam0_cam_2* R0_rect*Tr_velo_to_cam*Tr_imu_to_velo;
Pose = the ground truth pose in the IMU Coordinate System;

The pose_gt.txt in your demo is got by A*Pose*inv(A), I think it is not correct. 
The correct  pose_gt.txt should be got by B*Pose*inv(B) because the image in the paper is the image_02 and the image_02 is rectified.

meqdadhn, 2023-11-03:
> Hi @halajun
> 
> if: A = Tr_velo_to_cam×Tr_imu_to_velo; B = Tr_cam0_cam2× R0_rect×Tr_velo_to_cam×Tr_imu_to_velo; Pose = the ground truth pose in the IMU Coordinate System;
> 
> The pose_gt.txt in your demo is got by A×Pose×inv(A), I think it is not correct. The correct pose_gt.txt should be got by B×Pose×inv(B) because the image in the paper is the image_02 and the image_02 is rectified.

I believe what @SweetZed  mentioned is valid. Based on my tests, you guys assumed camera_2 rectified is the same as reference camera 0 (not rectified). Given that there is small translation between the two camera and also small angles between the origianal and rectified Camera 0, the generated ground truth for cam_pose.txt would be different from the real values specially as the camera goes further away from the origin. I beilive since only relative accuracy has been reported in your paper, this will not impact the final results too much, but it would be good if you could clarify this and maybe correct it in the next PR if possible. 

Thank you for your work on this, pretty interesting.

feiqingzhishang, 2024-12-16:
> Hi @halajun
>
> if: A = Tr_velo_to_cam×Tr_imu_to_velo; B = Tr_cam0_cam2× R0_rect×Tr_velo_to_cam×Tr_imu_to_velo; Pose = the ground truth pose in the IMU Coordinate System;
>
> The pose_gt.txt in your demo [...]

Hello, in B = Tr_cam0_cam2× R0_rect×Tr_velo_to_cam×Tr_imu_to_velo, the other three data have been obtained. How should R0_rect be obtained from calib?

feiqingzhishang, 2024-12-16:
> if: A = Tr_velo_to_cam×Tr_imu_to_velo; B = Tr_cam0_cam2× R0_rect×Tr_velo_to_cam×Tr_imu_to_velo; Pose = the ground truth pose in the IMU Coordinate System;
>
> The pose_gt.txt in your demo is got by [...]

Hello, in B = Tr_cam0_cam2× R0_rect×Tr_velo_to_cam×Tr_imu_to_velo, the other three data have been obtained. How should R0_rect be obtained from calib?

feiqingzhishang, 2024-12-16:
> Hi @halajun
>
> if: A = Tr_velo_to_cam×Tr_imu_to_velo; B = Tr_cam0_cam2× R0_rect×Tr_velo_to_cam×Tr_imu_to_velo; Pose = the ground truth pose in the IMU Coordinate System;
>
> The pose_gt.txt in your demo [...]
@SweetZed
Hello, in B = Tr_cam0_cam2× R0_rect×Tr_velo_to_cam×Tr_imu_to_velo, the other three data have been obtained. How should R0_rect be obtained from calib?

feiqingzhishang, 2024-12-16:
>
> 
> if: A = Tr_velo_to_cam×Tr_imu_to_velo; B = Tr_cam0_cam2× R0_rect×Tr_velo_to_cam×Tr_imu_to_velo; Pose = the ground truth pose in the IMU Coordinate System;
> @SweetZed
> The pose_gt.txt in your demo is got by A×Pose×inv(A), I think it is not correct. The correct pose_gt.txt should be got by B×Pose×inv(B) because the image in the paper is the image_02 and the image_02 is rectified.