halajun/VDO_SLAM #37: Problem about generating GT

gao19970315, 2022-01-27:
Hello. Thank you so much for your work, it's very cool.
I find the sequences of your demo. I use Matlab to generate the ground truth camera pose from the "data_tracking_oxts", the KITTI Tracking Dataset. 
I have some problems with GT generation. The GT I generate using Matlab has a larger error than the GT you generate.
e.g.
my GT:
143 0.988913918 0.000807471 -0.148487746 9.974954671 -0.002023517 0.999965642 -0.008038644 2.741625980 0.148476154 0.008249994 0.988881575 172.391896386 0.000000000 0.000000000 0.000000000 1.000000000
GT in kitti- demo：
143 0.988933274 -0.001158800 -0.148356442 -8.665262612 -0.000368661 0.999947215 -0.010267974 -0.690116978 0.148360527 0.010209034 0.988880646 172.520788973 0.000000000 0.000000000 0.000000000 1.000000000

When evo is used for evaluation, the rotation error is similar, but the translation error will increase.
And could you tell me how you extract it for details?
Finally, thank you again

gao19970315, 2022-04-24:
经过测试，找到了最终答案：
A=(R|T)_velo_to_cam * (R|T)_imu_to_velo，相机坐标系下的位姿变化T=A*pose{i}*inv(A)，此位姿与作者提供的pose_gt.txt完全一致。
具体的公式推导，后面更新

closed 2022-04-24 by gao19970315

darkyrs, 2022-05-19:
> 经过测试，找到了最终答案： A=(R|T)_velo_to_cam * (R|T)_imu_to_velo，相机坐标系下的位姿变化T=A*pose{i}*inv(A)，此位姿与作者提供的pose_gt.txt完全一致。 具体的公式推导，后面更新

大佬你好，我对Matlab不太熟悉，我想请教一下怎样用raw data中提供的devkit把data_tracking_oxts中的数据转换成pose_gt的格式呢？