halajun/VDO_SLAM #38: Problem when running demo-omd

yxianzhe, 2022-03-23:
Hi, this work is really fantastic.
When I tried demo-kitti, it worked well and got a very good estimation. I just had some problem when running demo-omd. And I had the following mistake:

..........Dealing with Camera Pose..........
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) /home/yxz/lib_sources/opencv/opencv/modules/calib3d/src/solvepnp.cpp:216: error: (-215:Assertion failed) npoints >= 4 && npoints == std::max(ipoints.checkVector(2, CV_32F), ipoints.checkVector(2, CV_64F)) in function 'solvePnPRansac'

I tried to modify the variable reprojectionError to have more good matches, but it didn't work. Do you have any ideas about this mistake?

ZhenhuiL1n, 2022-07-17:
I also have this problem.

RGthesis, 2022-09-01:
Hi, I am a begginer in this project and i am facing the same issue as you @yxianzhe @ZhenhuiL1n . If you have found a solution, let me know ;)

JonnyG123789, 2022-12-12:
it seems like too few  matched keypoints

JonnyG123789, 2022-12-13:
> Hi, I am a begginer in this project and i am facing the same issue as you @yxianzhe @ZhenhuiL1n . If you have found a solution, let me know ;)

DId you got this problem?

JonnyG123789, 2022-12-13:
The method is to  "#" about depth value in Tracking.cc 200.

Sherryblues, 2023-07-28:
> The method is to "#" about depth value in Tracking.cc 200.
hello!
I found that line 200 in Tracking.cc has already been annotated.
Can you explain it in more detail？
Thank you!
[image]

xuehu2, 2023-10-24:
> > The method is to "#" about depth value in Tracking.cc 200.
> > hello!
> > I found that line 200 in Tracking.cc has already been annotated.
> > Can you explain it in more detail？
> > Thank you!
> > [image]

if we use omd dataset, it is based on the RGB-D camera. So we should change the depth calculate function of RGB-D camera, such as line 193. 
so uncomment 193, and comment 191 can fix the problem.