halajun/VDO_SLAM #42: 关于OMD数据集的mask的提取

JonnyG123789, 2022-07-14:
作者您好，请问一下，你们在对目标位姿进行评估的时候，是如何从OMD上提取实例分割的mask对象的，能否把相关工具或者是代码发我一份，谢谢！1250131661@qq.com

zj2013, 2022-08-03:
tools文件夹里有的，是一个matlab代码。
[image]

JonnyG123789, 2022-08-03:
十分感谢！！！

zj2013, 2022-08-03:
同学你好，不知你有没有看到过OMD数据集的提供的深度图，我生成点云看了一下，误差大的根本用不了。墙面、标定板位置对应的点云确是波浪形。 不知作者是如何处理这种情况的。

JonnyG123789, 2022-08-03:
我们还没有在OMD数据集进行试验 但是三维激光生成的点云信息本来就是波浪形的吗？ 你可能需要转换以下。

JonnyG123789, 2022-08-03:
此外，还想请教一个问题，我想采用你们的方法去求解场景中的对象的位姿，但是在在利用前后帧匹配上的特征点对位姿进行求解(PNPRANSAC)的时候，发现内点inliers数量很少，在前后帧匹配上的特征点大概有2000左右 但是求解的位姿的内点只有20个左右 ，甚至更少 ，虽然我还没有进一步对求出来的位姿与真实位姿进行对比，但是我猜测这么少内点的情况下，这个求出来的位姿精度肯定不高。你知道这是什么原因造成的吗？而且我发现你代码中还没有进行特征匹配，求出的位姿的内点数都有几百个。