halajun/VDO_SLAM #43: how to get depth.png in kitti

andersonhusky, 2022-08-25:
I have a quesion about how you get the depth.png under the /demo-kitti/depth?

MinaHenein, 2022-08-25:
Hi @andersonhusky, This is explained in section IV. System in our paper VDO-SLAM: A Visual Dynamic Object-aware
SLAM System (https://arxiv.org/pdf/2005.11052.pdf)

andersonhusky, 2022-08-25:
thans for reply @MinaHenein! I know you are using MonoDepth2 to create depth image in Monocular mode, but it seems like the depth.png under the /demo-kitti/depth is calculated from Stereo mode

MinaHenein, 2022-08-25:
@andersonhusky Yes, correct.
From our paper:

The input to the system is stereo or RGB-D images. For stereo images, as a first step, we extract depth information by applying the stereo depth estimation method described in [62] to generate depth maps and the resulting data is treated as RGB-D.

[62] K. Yamaguchi, D. McAllester, and R. Urtasun, “Efficient Joint Segmentation, Occlusion Labeling, Stereo and Flow Estimation,” in European Conference on Computer Vision (ECCV). Springer, 2014, pp. 756–771.

andersonhusky, 2022-08-25:
got it! thanks again!