halajun/VDO_SLAM #51: unable to build using Dockerfile

Rong-Tao, 2023-08-08:
I downloaded the Dockfile provided and try to build using:
`docker build -t vdo_slam .`

And during the installation of 
`https://github.com/opencv/opencv_contrib/archive/3.4.0.zip `
This error occurs:
`fatal error: opencv2/xfeatures2d/cuda.hpp: No such file or directory`

 I think this is the problem of openCV extra modules, could you give an updated Dockfile such that it can compile?

xuehu2, 2023-09-09:
I have meet the same problem, and did u solve it?

Rong-Tao, 2023-09-11:
Well, this is relate to Docker having different port, so I you are in China and need a VPN for some of the sources, the docker won't work for you. I didn't manage to solve the proxy for host machine, docker and my VPN setting which is an absolute mess and headache. I just installed it on my host machine instead. PAIN

xuehu2, 2023-09-13:
Yes, I found that when I build the opencv_contrib lib from the source code, some packages could not been download. so I also try to install the enviroments in my own pc. not docker. And I also agree with the docker have some issue in China without a useful VPN. Thanks!

Rong-Tao, 2023-09-13:
It is not really the issue of VPN only, but it is extremely painful to setup Docker proxy while setting up your own laptop's proxy and somehow connect them. Because for your host machine, the ip is 127.0.0.1 which is the same for docker. The docker machine think itself is 127.0.0.1 and your host machine is sth else, load of headache is ahead.

xuehu2, 2023-09-14:
And I have a question, can I build a clean image of ubuntu16.04, then run this pure image on docker, and get into the image's terminal(but Idk the way to use the image's terminal.), to modify some settings to help the mirror machine install these packages?

Rong-Tao, 2023-09-15:
yes, you can 
1. either setup proper proxy first on your host machine then your docker image
2. You can pre download the source needed on your host machine (The missing files in the error msg) and include them in the build process, I think is like 20-30 ish