halajun/VDO_SLAM #53: Segment error (core dumped)

Alwen-Yan-edu, 2023-09-09:
I admire your excellent work very much. I have a question that urgently needs your answer: After I successfully compiled the executable file vdo_slam, the terminal immediately reported "Segment error (core dumped)" after I entered the instruction. This is different from orb-slam series, which reports an error immediately without starting to run.
My environment is:
ubuntu18.04
opencv3.4.0
your dependencies/g2o
VTK6.3.0
Ceres1.13.0
[image]

Alwen-Yan-edu, 2023-09-11:
[image]

xuehu2, 2023-10-24:
I had met this error. It may be the ‘CSPARSE’ in the end of cmake_modules should be 'CSparse'. But it is long time ago, I suggest you try it, maybe can fix it.

closed 2023-10-24 by Alwen-V