halajun/VDO_SLAM #54: 为什么这样计算深度值？

theyunfeng, 2023-09-12:
[image]
为何RGBD和双目的深度计算公式一样且在分母中除mDepthMapFactor？

Sherryblues, 2023-10-03:
https://blog.csdn.net/catpico/article/details/120688795

Sherryblues, 2023-10-03:
https://blog.csdn.net/catpico/article/details/120688795
我更好奇为什么用mbf来除以深度值，mbf应该是双目基线长度乘以fx，
这个深度值在后续用来判断远点和近点，那么也应该是用深度值除以基线长度，得到的倍数和远点近点阈值作比较更合理一点吧

Cai-RS, 2023-10-07:
> https://blog.csdn.net/catpico/article/details/120688795 我更好奇为什么用mbf来除以深度值，mbf应该是双目基线长度乘以fx， 这个深度值在后续用来判断远点和近点，那么也应该是用深度值除以基线长度，得到的倍数和远点近点阈值作比较更合理一点吧

双目的原始depth图（等式右边的imD)其实存储的是左右相机的视差，这里的计算就是对应了双目深度的计算公式，mDepthMapFactor这个值应该代表的是焦距f（当然可能对真实数据还有别的比例缩放，具体看原始imD的计算方式）。

Sherryblues, 2023-10-08:
> > https://blog.csdn.net/catpico/article/details/120688795 我更好奇为什么用mbf来除以深度值，mbf应该是双目基线长度乘以fx， 这个深度值在后续用来判断远点和近点，那么也应该是用深度值除以基线长度，得到的倍数和远点近点阈值作比较更合理一点吧
> 
> 双目的原始depth图（等式右边的imD)其实存储的是左右相机的视差，这里的计算就是对应了双目深度的计算公式，mDepthMapFactor这个值应该代表的是焦距f（当然可能对真实数据还有别的比例缩放，具体看原始imD的计算方式）。

好的好的，谢谢，我去看看深度计算那边

theyunfeng, 2023-10-13:
> > https://blog.csdn.net/catpico/article/details/120688795 我更好奇为什么用mbf来除以深度值，mbf应该是双目基线长度乘以fx， 这个深度值在后续用来判断远点和近点，那么也应该是用深度值除以基线长度，得到的倍数和远点近点阈值作比较更合理一点吧
> 
> 双目的原始depth图（等式右边的imD)其实存储的是左右相机的视差，这里的计算就是对应了双目深度的计算公式，mDepthMapFactor这个值应该代表的是焦距f（当然可能对真实数据还有别的比例缩放，具体看原始imD的计算方式）。

但是他这里如果深度图是RGBD的深度图的话，还用双目的计算公式明显存在问题

Cai-RS, 2023-10-13:
> 但是他这里如果深度图是RGBD的深度图的话，还用双目的计算公式明显存在问题

你仔细看一下代码？针对双目和RGBD这里的计算公式不一样，RGBD的计算甚至直接给注释掉了，要么是在别处进行/mDepthMapFactor的操作，要么是RGBD的原始深度就是真实比例，不存在缩放。这都跟原始Depthmap的数据来源相关。