halajun/VDO_SLAM #55: 光流值的计算方式

Cai-RS, 2023-09-17:
作者你好，感谢你开源了自己的工作，让我们后辈们可以学习！我在阅读你的代码时，有一个疑问：你用网络计算的光流表示的是 （后一帧中点位置 - 前一帧中对应点位置）还是反过来的？我看你的代码，似乎默认光流值是 下一帧点位置-当前帧点位置。但这就产生了问题，你把当前帧的RGB图像与光流图像一起作为输入来处理，但是光流图本身又得有下一帧图像参与才能得到，这貌似违背了因果吧？还是说默认你的系统当前处理的图像总是比最新的图像早一帧？

Sherryblues, 2023-10-03:
KITTI的y轴是垂直地面的，z轴向前

Sherryblues, 2023-10-03:
KITTI的y轴是垂直地面朝下的，z轴朝前

Cai-RS, 2023-10-07:
> KITTI的y轴是垂直地面朝下的，z轴朝前

原来是这样，感谢~

xuehu2, 2023-10-26:
> 作者你好，感谢你开源了自己的工作，让我们后辈们可以学习！我在阅读你的代码时，有一个疑问：你用网络计算的光流表示的是 （后一帧中点位置 - 前一帧中对应点位置）还是反过来的？我看你的代码，似乎默认光流值是 下一帧点位置-当前帧点位置。但这就产生了问题，你把当前帧的RGB图像与光流图像一起作为输入来处理，但是光流图本身又得有下一帧图像参与才能得到，这貌似违背了因果吧？还是说默认你的系统当前处理的图像总是比最新的图像早一帧？

你好，请问你知道他为何这样反过来操作吗？

Cai-RS, 2023-10-26:
读完了论文和代码，直接说结论吧，此VDO系统并不是实时的，它是先对给定的原始图像序列计算逐帧的深度、光流和语义map，然后再把这些和原始图像序列逐帧地给到系统中进行处理。至于光流值，则是 后一帧中的点坐标 - 前一帧中的点坐标，也就是说，系统当前处理某一帧时可以提前计算出下一帧中的匹配点位置。整个系统本身无法实时，所以这种定义和处理的方式就无所谓了。